roboticslab-uc3m/speech #2: Problems finded with Speech_IDL

rsantos88, 2017-03-15:
I found the following problems:
When I am trying to compile `espeakExample`. it shows this message:
```bash 
[ 50%] Building CXX object CMakeFiles/espeakExample.dir/EspeakExample.cpp.o
[100%] Building CXX object CMakeFiles/espeakExample.dir/main.cpp.o
Linking CXX executable espeakExample
CMakeFiles/espeakExample.dir/EspeakExample.cpp.o: En la función `bool yarp::dev::DeviceDriver::view<Speech_IDL>(Speech_IDL*&)':
EspeakExample.cpp:(.text._ZN4yarp3dev12DeviceDriver4viewI10Speech_IDLEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewI10Speech_IDLEEbRPT_]+0x46): referencia a `typeinfo for Speech_IDL' sin definir
collect2: error: ld returned 1 exit status
CMakeFiles/espeakExample.dir/build.make:116: recipe for target 'espeakExample' failed
make[2]: *** [espeakExample] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/espeakExample.dir/all' failed
make[1]: *** [CMakeFiles/espeakExample.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
```
@jgvictores saw something estrange...
`Speech_IDL.h` should have a virtual methods "pure virtual" like `virtual void Method() = 0;`
This can explain the previous error. I leave an interesting explanation of this:
>> If the class contains any non-pure virtual functions, then the compiler will assume that they have an implementation somewhere, and its internal structures (vtable and typeinfo) might be generated in the same object file as one of those; if those functions are not implemented, then the internal structures will be missing and you will get these.

So, we have to resolve this...  :confounded:

jgvictores, 2017-03-15:
Taking another look at it, I think we can touch our `.thrift` to add the `=0`.

jgvictores, 2017-03-17:
Reference https://thrift.apache.org/docs/types mentions `pure virtual abstract class`. Must check this more in depth.

jgvictores, 2017-04-11:
While our own example looks good and complies with yarpdev mechanisms (namely, `view`), http://yarp.it/thrift_tutorial_simple.html shows a different use.

PeterBowman, 2017-07-13:
> Speech_IDL.h should have a virtual methods "pure virtual" like virtual void Method() = 0;

Maybe it shouldn't? The tutorial itself states that CMake macros generate a .cpp implementation file ([Step Two: Source code generation](http://yarp.it/thrift_tutorial_simple.html)). This and the lack of a `= 0` would clearly indicate why the linker step crashes: it's unable to find an actual implementation of the `Espeak` class and the symbols its `.cpp.o` file would export - the compiler didn't even create one. I managed to compile successfully after replacing [example/cpp/espeakExample/CMakeLists.txt#L32](https://github.com/roboticslab-uc3m/speech/blob/4110c70596ecd01ccb9786e16c2fe2f8b3fae1b9/example/cpp/espeakExample/CMakeLists.txt#L32) with:

```cmake
add_executable(${KEYWORD}
               ${folder_source}
               ${folder_header}
               "/path/to/speech/build/libraries/SpeechYarp/Espeak/src/Speech_IDL.cpp")
```

See also [robotology/yarp/example/idl/thrift/generated](https://github.com/robotology/yarp/tree/030e29d129509c036f59e160e39ac0ce53f28931/example/idl/thrift/generated), specifically the .cmake files since they set the `THRIFT_SOURCES` variable and provide it to the compiler at `add_library`.

PeterBowman, 2017-07-14:
Done at b2a6901.

jgvictores, 2017-08-29:
@PeterBowman Should we consider this a hack?
The original intention was something more like: http://yarp.it/thrift_tutorial_simple.html#thrift_tutorial_simple_sec_client

PeterBowman, 2017-08-29:
> @PeterBowman Should we consider this a hack?

Why? You really need to pass your .cpp's to a library or executable target, so I'd say this is the way it's supposed to be (cf. YARP tutorial & examples). My patch could be improved as stated in CMake comments, but the original issue (a mere case of build failure) should have been already addressed. Did this ever use to work before?

I didn't look for deeper insight into `EspeakExample` nor the sample client functionality from the tutorial (how does it relate to `EspeakExample`?), perhaps that could be investigated in a new issue? Apart from that, I don't think you can `view` a Thrift interface just like any other device (at least the tutorials don't say how).

PeterBowman, 2017-08-29:
> Reference https://thrift.apache.org/docs/types mentions pure virtual abstract class. Must check this more in depth.

I think that Thrift services simply *look like* pure virtual abstract classes as you don't need to worry about an implementation on the server part, but still the compiler generates one for you (plus stubs for the client side):

> Service definitions are semantically equivalent to defining an interface (or a pure virtual abstract class) in object-oriented programming. The Thrift compiler **generates** fully functional client and server stubs that **implement** the interface.

(https://diwakergupta.github.io/thrift-missing-guide/#_services)

> Think of service definitions as Java interfaces — you need to supply a name and signatures for the methods. Optionally, a service may extend other services.
> The Thrift compiler will **generate** service interface code (for the server) and stubs (for the client) in your chosen language. Thrift ships with RPC libraries for most languages that you can then use to run your client and server.

(https://diwakergupta.github.io/thrift-missing-guide/#_defining_services)

> For each service, a .h and a .cpp file are created, which contain the definition of the interface as a class derived from yarp::os::Wire.

(http://yarp.it/thrift_tutorial.html (#Services))

jgvictores, 2017-08-29:
Yes, this wasn't explicit at all. Started #9.

commit a88e42ea (2017-07-14): Compile example program with ROBOTICSLAB_SPEECH_IDL_FILES

Actually fixes #2.
closed 2017-08-29 by jgvictores