roboticslab-uc3m/speech #4: Think about gorinars/ros_voice_control

jgvictores, 2017-03-28:
Think about https://github.com/gorinars/ros_voice_control
- uses [pocketsphinx-python](https://github.com/cmusphinx/pocketsphinx-python) **instead of Gstreamer**
- enables the keyword search mode, which should better filter not needed words
- is really simple (just a script to run)

jgvictores, 2017-03-31:
Issue refers to current  [speechRecognition.py](https://github.com/roboticslab-uc3m/speech/blob/c319da5c21aa2bd7886133a9e5dc3b8bdf24c0d8/programs/speechRecognition/speechRecognition.py).

PeterBowman, 2021-11-14:
We are happy with the current state of speechRecognition.py, see https://github.com/roboticslab-uc3m/speech/issues/8#issuecomment-968270204.

closed 2022-05-23 by PeterBowman