roboticslab-uc3m/speech #6: Add tests

jgvictores, 2017-04-11:
Add tests (e.g. via gtest).

jgvictores, 2017-04-11:
For this repo, python tests also make sense.

jgvictores, 2017-04-11:
Blocked by #2.

jgvictores, 2017-12-07:
Now #2 is closed, but blocked by #9.

PeterBowman, 2021-11-11:
Actually, #9 is not the way to proceed since we want to instantiate an Espeak device locally. However, our current Espeak device always opens a YARP server port, which is not desirable in the context of unit testing. I'd rather move the implementation into a new shared library. After doing that, I think it would make more sense to serve Espeak RPC commands through an RFModule's port instead of via yarpdev.

PeterBowman, 2021-11-11:
> I'd rather move the implementation into a new shared library. After doing that, I think it would make more sense to serve Espeak RPC commands through an RFModule's port instead of via yarpdev.

Done at https://github.com/roboticslab-uc3m/speech/commit/72c6e46887cd5f006d91f6681446806817255ae0, the name of the new RFModule-based app is `espeakServer`.

Tests added at https://github.com/roboticslab-uc3m/speech/commit/414e77177c734d0d78e535062d41819872b03a8d. I had to struggle with ALSA/PulseAudio due to a missing sound device on the runner. [This solution](https://stackoverflow.com/a/40095791/10404307) worked nicely on Ubuntu Bionic, but not on Focal. I'm explicitly starting the PA daemon at https://github.com/roboticslab-uc3m/speech/commit/2bdd9db6c3f5523703e793c1c0ff05e211194866 in order to overcome that, and also adding a small delay in the test suite seemed to be necessary (not entirely sure that it won't fail again). As a side note, `modprobe snd_dummy` is pointless, there is no such module.

Merged at https://github.com/roboticslab-uc3m/speech/commit/7633b8c07251a5cd3f4af5a29e3e2039d05c55d6.

closed 2021-11-11 by PeterBowman