roboticslab-uc3m/speech #8: [idea]  Publish microphone data stream

jgvictores, 2017-04-22:
_From @jgvictores on November 28, 2015 11:52_

Had issues last time using http://wiki.icub.org/wiki/Grabbing_sound_with_PortAudio.

_Copied from original issue: roboticslab-uc3m/vision#5_

davidvelascogarcia, 2019-07-04:
I was doing some test with `portaudio` in this [repo](https://github.com/davidvelascogarcia/yarp-streaming-audio/blob/master/programs/linux/yarpStreamingAudio/sender/main.cpp). [Installation guides](http://portaudio.com/docs/v19-doxydocs/compile_linux.html
) that i use. Also i added `yarp device portaudio`. Maybe it could serve.

PeterBowman, 2019-07-04:
The portaudio device has been deprecated in [YARP 3.2.0](https://github.com/robotology/yarp/releases/tag/v3.2.0) and replaced by [portaudioPlayer](http://www.yarp.it/classPortAudioRecorderDeviceDriver.html), [portaudioRecorder](http://www.yarp.it/classPortAudioPlayerDeviceDriver.html).

PeterBowman, 2021-11-13:
Should be doable, see [Audio in YARP](http://www.yarp.it/latest/AudioDoc.html). What do wee need this for, @jgvictores?

jgvictores, 2021-11-14:
The rationale behind this proposal was in line with [CBSE](https://en.wikipedia.org/wiki/Component-based_software_engineering) practices. We have a monolithic data acquisition and processing approach in our ASR, which seems to go against CBSE.

However, in this case, I would close this issue as `invalid`/`wontfix`. I can only identify 2 benefits of decoupling, which are not critical in our case:
1. Make the ASR microphone-agnostic -> Not required, as the current implementation is relatively microphone-agnostic.
2. Outsource the ASR computational cost -> Not required, as the current ASR computational cost is not very high.

PS1: Similar logic applies to our TTS, where we do not decouple at the generated signal level, nor does this seem very necessary.
PS2: Additionally, I've played around with YARP audio devices, and (possibly due to how audio is managed at OS level) it was relatively tricky to get it to work without too much latency.
PS3: https://icub-tech-iit.github.io/documentation/icub_microphones/icub_microphones/ is also very nice.

closed 2021-11-14 by jgvictores

PeterBowman, 2024-08-15:
We might revisit this in https://github.com/roboticslab-uc3m/speech/issues/31.