roboticslab-uc3m/speech #10: Document ASR and TTS

Kuronogard, 2017-09-05:
Create documentation for ASR and TTS usage.

jgvictores, 2017-09-05:
- ASR (`speechRecognition.py`):
  - Progress up to 71d32a9a28dd6620792add01feab700e0e0ece13 can be seen in http://robots.uc3m.es/dox-speech/group__speechRecognition.html. 
  - Must document its RPC port.

- TTS (`yarpdev --device Espeak`):
  - Must do.

jgvictores, 2017-09-09:
Still missing documentation, but already some at:
- ASR: http://robots.uc3m.es/dox-speech/group__speechRecognition.html
- TTS: http://robots.uc3m.es/dox-speech/group__Espeak.html

rsantos88, 2018-01-11:
@jgvictores can I help you with this?

jgvictores, 2018-01-11:
Sure, thanks!
Basically, the documentation of [speechRecognition.py](https://github.com/roboticslab-uc3m/speech/blob/a304107d88ac7ea244804d4da6c2caac57091ad7/programs/speechRecognition/speechRecognition.py) and [Espeak](https://github.com/roboticslab-uc3m/speech/tree/a304107d88ac7ea244804d4da6c2caac57091ad7/libraries/SpeechYarp/Espeak) is pretty poor. Documentation should be improved, e.g.:
- How to launch
- What parameters there are
- Improve/create examples
- Link to examples

Blocked by #14

PeterBowman, 2018-01-12:
#14 done, unblocking.

rsantos88, 2018-01-17:
That is finish. You can see the result [here](https://github.com/roboticslab-uc3m/speech/blob/develop/doc/speech-instructions.md)
If you see something wrong, please tell me or feel free to correct it :grin:

jgvictores, 2018-01-18:
@rsantos88 Looks really good! Thanks a lot for all of this.

I'm leaving it open because I still want to link a few things together, but thanks a lot for all of this useful and very helpful information.

jgvictores, 2018-01-20:
Okay, I've reviewed this again, I think it's perfect.

Only objections are related to practices that, if ever defined, will propagate through all repositories. Related issue already points to here: https://github.com/roboticslab-uc3m/questions-and-answers/issues/29

@rsantos88 Feel free to close this issue (with http://robots.uc3m.es/contributors in mind xD)!

closed 2018-01-22 by rsantos88