roboticslab-uc3m/speech #13: Research on muting microphone

jgvictores, 2017-11-20:
Research on muting microphone. Several Python multiplatform solutions, but right now need c++ on Ubuntu. Here are some references. I believe we are using ALSA, not pulseaudio, so will check first option.
1. https://askubuntu.com/questions/26068/how-do-you-mute-from-the-command-line
1. https://stackoverflow.com/questions/3985014/linux-alsa-sound-api-questions-how-do-you-mute

jgvictores, 2017-11-20:
Better:
```bash
amixer -c 1 sset Mic nocap
amixer -c 1 sset Mic cap
```
Ref:
- https://askubuntu.com/questions/12100/command-to-mute-and-unmute-a-microphone/13364

rsantos88, 2018-01-16:
I've been checking this lines but it doesn't work. When I put this line, it shows:
```bash 
amixer: Unable to find simple control 'Mic',0
```
So, I've tried to put my control, checking first with  `amixer scontrols`
```bash
Simple mixer control 'IEC958',0
```
but it doesn't recognize `nocap` or `cap` command:
```bash
amixer: Invalid command!
```
Other option is using `alsamixer` but as indicated in [this discussion](https://ubuntuforums.org/showthread.php?t=840573) it's not working when we try to muting/turning down the mic. The only way that I found for mute it is changing the `Input Source` to another that is not connected (for example `Rear Mic`)

jgvictores, 2018-01-16:
Okay, maybe this is pretty platform-specific and on future comments we should document on which system (PC, distro, version) each specific command was working. :sweat_smile:

rsantos88, 2018-04-04:
From tiago-main `t1-barcelona-microphone` branch:
```cpp
	    res = system ("amixer -c 1 sset Mic cap");
        cout << "--Microphone ON--" << endl;

		while (!speechRecognitionPort.read(bot, false));

	    res = system ("amixer -c 1 sset Mic nocap");
        cout << "--Microphone OFF--" << endl;
```

rsantos88, 2018-04-05:
Project https://github.com/larsimmisch/pyalsaaudio looks interesting for `speechRecognition.py`. Can be installed easily:
```
sudo apt install libasound2-dev
pip install pyalsaaudio --user
```
And run example `mixertest.py`.

rsantos88, 2018-04-05:
Possible idea for mute the microphone:
```shell
* [terminal 1] speechRecognition.py
* [terminal 2] yarp rpc /speechRecognition/rpc:s
>> set mute 
>> set unmute
```
using this port allow us to mute the microphone running the demostration (follow-me, waiter...) when it is necessary, e.g when the robot is speaking or doing some action and we want to avoid the noise.

rsantos88, 2018-04-11:
implemented and working at (#18)

rsantos88, 2018-04-11:
**ToDo:** to adapt the demostrations with this functionality (follow-me and teo-waiter)

rsantos88, 2018-04-11:
**summary to do:**
- [ ] Prepare `teo-head` to use microphone bluetooth
- [ ] Document installation `pyalsaaudio` packets
- [ ] Document possible errors with bluetooth microphone
- [ ] Prepare demostrations  to use mute/unmute functionality

rsantos88, 2018-04-12:
> - [ ] Prepare `teo-head` to use microphone bluetooth

At this point, I found that `teo-head` incluse a internal bluetooth receiver. This is recognize and automatically activated by the O.S but it has a very poor coverage. So, I want to add the `usb-bluetooth`. I need to know how to disable the internal bluetooth and enable only the `usb-bluetooth` detection.

rsantos88, 2018-04-12:
To show a list of this devices:
`rfkill list all`

rsantos88, 2018-04-13:
> - [x] Prepare `teo-head` to use microphone bluetooth
>
>At this point, I found that `teo-head` include a internal bluetooth receiver. This is recognized and automatically activated by the O.S but it has a very poor coverage. So, I want to add the `usb-bluetooth`. I need to know how to disable the internal bluetooth and enable only the `usb-bluetooth` detection. 

I resolved this problem. I'll put here the steps:
1. See the devices. Should be some on the Intel devices.  
In this case : **Bus 003 Device 003: ID 8087:07da Intel Corp**
```bash
lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046a:0010 Cherry GmbH SmartBoard XX44
Bus 003 Device 003: ID 8087:07da Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
2. Disable this device here:
```bash
sudo nano /etc/udev/rules.d/81-bluetooth-hci.rules
```
and pasting this 
```bash
SUBSYSTEM=="usb", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="07da", ATTR{authorized}="0"
```
Save, exit nano and reboot

PeterBowman, 2018-04-17:
One thing I can come up with are acknowledgements. That is, make TEO send an *Are you sure?* notice back to you, and await confirmation. Is it feasible in you demo workflow?

rsantos88, 2018-04-17:
> One thing I can come up with are acknowledgements. That is, make TEO send an Are you sure? notice back to you, and await confirmation. Is it feasible in your demo workflow?

Yes, it's feasible but in practice, Teo would be all the time trying to await confirmation or receiving wrong inputs.

rsantos88, 2018-04-17:
Other possible idea is to filter the noise with greater concatenation of words: E.g `please stop TEO` instead of `stop` but I want to get a solution to avoid modifying the dictionaries...

rsantos88, 2018-04-19:
Note: to restart audio devices `sudo /etc/init.d/pulseaudio restart`

rsantos88, 2018-10-08:
related: https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/29

jgvictores, 2018-12-13:
@rsantos88 What should be do with this issue?
1. Consider it blocked by https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/29 ?
2. Leave it open until we document the conclusions/recommendations at https://github.com/roboticslab-uc3m/developer-manual ?

closed 2021-11-14 by PeterBowman

PeterBowman, 2023-06-29:
Another take on this through Python's sounddevice package: 2dd945939c7142d374912b4a5a5dee130724872c.