roboticslab-uc3m/speech #16: The dictionary change is not working

rsantos88, 2018-01-17:
When I change to another dictionary, e.g: `setDictionary waiter spanish`, the program `speechRecognition` freezes.
I suppose seeing the bash that the problem is that it doesn't load the model correctly...
```bash
Invalid Region Code.
||| finding file dictionary/follow-me-spanish.lm
||| checking [/home/teo/repos/speech/programs/speechRecognition/dictionary/follow-me-spanish.lm] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/follow-me/robots/default] (robot yarp.d)
||| checking [/usr/local/share/roboticslab-speech/robots/default] (robot yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-configuration-files/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-grasp/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-openrave-models/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-self-presentation/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/roboticslab-speech/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/roboticslab-speech/contexts/speechRecognition
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-configuration-files/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-grasp/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-openrave-models/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-self-presentation/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.lm] (context)
||| found /usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.lm
||| finding file dictionary/follow-me-spanish.dic
||| checking [/home/teo/repos/speech/programs/speechRecognition/dictionary/follow-me-spanish.dic] (pwd)
||| checking [/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.dic] (context)
||| found /usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.dic
||| finding file model/
||| checking [/home/teo/repos/speech/programs/speechRecognition/model/] (pwd)
||| checking [/usr/local/share/roboticslab-speech/contexts/speechRecognition/model/] (context)
||| found /usr/local/share/roboticslab-speech/contexts/speechRecognition/model/
Dictionary changed successfully (/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.lm) (/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.dic) (/usr/local/share/roboticslab-speech/contexts/speechRecognition/model/)
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2,000000e+00
-allphone				
-allphone_ci		yes		yes
-alpha			0.97		9,700000e-01
-ascale			20.0		2,000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1,000000e-20
-bestpath		yes		yes
-bestpathlw		9.5		9,500000e+00
-ceplen			13		13
-cmn			live		live
-cmninit		40,3,-1		40,3,-1
-compallsen		no		no
-dict					/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.dic
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1,000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1,000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8,500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7,000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/share/roboticslab-speech/contexts/speechRecognition/model/
-input_endian		little		little
-jsgf					
-keyphrase				
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1,000000e-01
-kws_threshold		1e-30		1,000000e-30
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		0
-lm					/usr/local/share/roboticslab-speech/contexts/speechRecognition/dictionary/follow-me-spanish.lm
-lmctl					
-lmname					
-logbase		1.0001		1,000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1,333333e+02
-lpbeam			1e-40		1,000000e-40
-lponlybeam		7e-29		7,000000e-29
-lw			6.5		6,500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1,000000e-07
-mllr					
-mmap			yes		yes
-ncep			13		13
-nfft			512		512
-nfilt			40		40
-nwpen			1.0		1,000000e+00
-pbeam			1e-48		1,000000e-48
-pip			1.0		1,000000e+00
-pl_beam		1e-10		1,000000e-10
-pl_pbeam		1e-10		1,000000e-10
-pl_pip			1.0		1,000000e+00
-pl_weight		3.0		3,000000e+00
-pl_window		5		5
-rawlogdir				
-remove_dc		no		no
-remove_noise		yes		yes
-remove_silence		yes		yes
-round_filters		yes		yes
-samprate		16000		1,600000e+04
-seed			-1		-1
-sendump				
-senlogdir				
-senmgau				
-silprob		0.005		5,000000e-03
-smoothspec		no		no
-svspec					
-tmat					
-tmatfloor		0.0001		1,000000e-04
-topn			4		4
-topn_beam		0		0
-toprule				
-transform		legacy		legacy
-unit_area		yes		yes
-upperf			6855.4976	6,855498e+03
-uw			1.0		1,000000e+00
-vad_postspeech		50		50
-vad_prespeech		20		20
-vad_startspeech	10		10
-vad_threshold		3.0		3,000000e+00
-var					
-varfloor		0.0001		1,000000e-04
-varnorm		no		no
-verbose		no		no
-warp_params				
-warp_type		inverse_linear	inverse_linear
-wbeam			7e-29		7,000000e-29
-wip			0.65		6,500000e-01
-wlen			0.025625	2,562500e-02

Dictionary changed to follow-me-spanish.lm, follow-me-spanish.dic,
```

rsantos88, 2018-01-17:
error fixed (cebca1843374b2f608aecc840ceb50f33f396a72)

closed 2018-01-17 by rsantos88

jgvictores, 2018-01-17:
Looks like a good fix, thanks @rsantos88 !!!

Kuronogard, 2018-01-17:
Ops, sorry, I was aware of the problem, but forgot completely about it... :S
Thanks for the fix, @rsantos88.