roboticslab-uc3m/speech #17: Refactor CMake files

PeterBowman, 2018-01-20:
* roboticslab-uc3m/kinematics-dynamics#14
* roboticslab-uc3m/yarp-devices#26
* roboticslab-uc3m/openrave-yarp-plugins#56
* roboticslab-uc3m/vision#7

PeterBowman, 2018-01-21:
Done at d80194c. See instructions and example on how to use IDL files in client apps:

* [cmake/template/ROBOTICSLAB_SPEECHConfig.cmake.in](https://github.com/roboticslab-uc3m/speech/blob/d80194cf1aea507112b3ec713d168618729c024b/cmake/template/ROBOTICSLAB_SPEECHConfig.cmake.in)
* [example/cpp/espeakExample/CMakeLists.txt](https://github.com/roboticslab-uc3m/speech/blob/d80194cf1aea507112b3ec713d168618729c024b/example/cpp/espeakExample/CMakeLists.txt)

closed 2018-01-21 by PeterBowman

jgvictores, 2018-01-21:
Cool! Thanks!!