roboticslab-uc3m/speech #22: speech-instructions.md are coupled

jgvictores, 2018-04-25:
https://github.com/roboticslab-uc3m/speech/blob/2ed13755d900721a139f300daf37ad185bc7fcf6/doc/speech-instructions.md should be moved into corresponding README files.

rsantos88, 2018-05-08:
So, do you want to pass this instructions directly to https://github.com/roboticslab-uc3m/speech/blob/develop/README.md?

jgvictores, 2018-05-08:
speechRecognition -> https://github.com/roboticslab-uc3m/speech/tree/develop/programs/speechRecognition/README.md

Espeak -> https://github.com/roboticslab-uc3m/speech/tree/develop/libraries/YarpPlugins/Espeak/README.md

rsantos88, 2018-05-08:
bad links

jgvictores, 2018-05-08:
Nope, files to be created. Links to the future.  : )

rsantos88, 2018-05-08:
aha! ok ok

rsantos88, 2018-05-08:
done https://github.com/roboticslab-uc3m/speech/pull/23

jgvictores, 2018-05-09:
Cool, closing! Thanks!!

closed 2018-05-09 by jgvictores