roboticslab-uc3m/speech #25: Repository is heavy

jgvictores, 2018-05-16:
Repository is heavy: `135 M` ATOW (0c5c94159201f15dbb2bd078412704e6bd0f08f0).

PeterBowman, 2018-05-16:
Most of which (130 MB) is used by the `share/speechRecognition/conf/` directory alone. Let's break this down by file:

```shell
$ find . -type f -exec ls -lhgGS "{}" \+
-rwxrwxrwx 1  67M Mar 31 22:45 ./dictionary/es-20k.lm
-rwxrwxrwx 1  23M Mar 31 22:45 ./model/en-us/means
-rwxrwxrwx 1  23M Mar 31 22:45 ./model/en-us/variances
-rwxrwxrwx 1 6.7M Mar 31 22:45 ./model/en-us/mdef
-rwxrwxrwx 1 6.0M Mar 31 22:45 ./model/es/mixture_weights
-rwxrwxrwx 1 1.5M Mar 31 22:45 ./model/es/sendump
-rwxrwxrwx 1 1.3M Mar 31 22:45 ./model/es/mdef
-rwxrwxrwx 1 641K Mar 31 22:45 ./model/en-us/mixture_weights
-rwxrwxrwx 1 588K Mar 31 22:45 ./dictionary/es-20k.dic
-rwxrwxrwx 1 508K Mar 31 22:45 ./model/es/means
-rwxrwxrwx 1 508K Mar 31 22:45 ./model/es/variances
-rwxrwxrwx 1 5.6K Mar 31 22:45 ./model/en-us/feature_transform
-rwxrwxrwx 1 2.1K Mar 31 22:45 ./model/en-us/transition_matrices
-rwxrwxrwx 1 1.6K Mar 31 22:45 ./model/en-us/README
-rwxrwxrwx 1 1.4K Mar 31 22:45 ./dictionary/follow-me-english.lm
-rwxrwxrwx 1 1.4K Mar 31 22:45 ./dictionary/waiter-english.lm
-rwxrwxrwx 1 1.3K Mar 31 22:45 ./model/es/transition_matrices
-rwxrwxrwx 1  554 Mar 31 22:45 ./dictionary/waiter-spanish.lm
-rwxrwxrwx 1  466 Mar 31 22:45 ./dictionary/follow-me-spanish.lm
-rwxrwxrwx 1  148 Mar 31 22:45 ./model/es/feat.params
-rwxrwxrwx 1  114 Mar 31 22:45 ./dictionary/follow-me-english.dic
-rwxrwxrwx 1  113 Mar 31 22:45 ./dictionary/waiter-english.dic
-rwxrwxrwx 1  112 Mar 31 22:45 ./model/en-us/feat.params
-rwxrwxrwx 1  108 Mar 31 22:45 ./dictionary/waiter-spanish.dic
-rwxrwxrwx 1   76 Mar 31 22:45 ./dictionary/follow-me-spanish.dic
-rwxrwxrwx 1   56 Mar 31 22:45 ./model/en-us/noisedict
-rwxrwxrwx 1   43 Mar 31 22:45 ./dictionary/follow-me-english.txt
-rwxrwxrwx 1   43 Mar 31 22:45 ./dictionary/waiter-english.txt
-rwxrwxrwx 1   42 Mar 31 22:45 ./dictionary/waiter-spanish.txt
-rwxrwxrwx 1   31 Mar 31 22:45 ./dictionary/follow-me-spanish.txt
-rwxrwxrwx 1   30 Mar 31 22:45 ./speechRecognition.ini
-rwxrwxrwx 1   27 Mar 31 22:45 ./model/es/noisedict
```

PeterBowman, 2021-11-14:
Related: https://github.com/roboticslab-uc3m/vision/issues/64 (in the past, speech and vision were coupled in a single teo-head repo).

A freshly clone today pulls 204 MB from GH, where 130 MB correspond to `share/` alone, and 74 MB to `.git/`. We are probably not moving speech files elsewhere, and I'm not especially keen on rewriting Git history via BFG anymore. I'd say the current size is bearable and let's move on.

jgvictores, 2021-11-14:
A small note (in case we ever do want more models/etc): DNN frameworks such as `keras` (I think I've also seen this with `pytorch`) in fact download "heavy" model files on first run, storing them in a file cache (e.g. under `~/.local` or similar in Linux).

But yes, for now the size is bearable. 👍

PeterBowman, 2021-11-14:
> DNN frameworks such as `keras` (I think I've also seen this with `pytorch`) in fact download "heavy" model files on first run, storing them in a file cache

Oh, we also do that at roboticslab-uc3m/vision, see [this CMakeLists.txt](https://github.com/roboticslab-uc3m/vision/blob/7b725f1a7150fd48e95cf585327b764ecbd097b5/models/CMakeLists.txt).

closed 2022-05-23 by PeterBowman